uf-mil/mil #1283: Create async `electrical_protocol` driver

cbrxyz, 2024-09-27:
### What needs to change?

Currently, we have a synchronous version of the serial driver used by `electrical_protocol`. However, it would be great to have an asynchronous version of this driver. This would empower us to combine our electrical drivers with other forms of I/O (including ROS nodes/topics/etc.).

This could be done by copying the current synchronous driver and replacing the synchronous parts with [`pyserial-asyncio`](https://github.com/pyserial/pyserial-asyncio) and [`asyncio`](https://docs.python.org/3/library/asyncio.html). I also made some brief starting progress on this task on [this branch](https://github.com/uf-mil/mil/tree/electrical-protocol-async).

### How would this task be tested?

1. Copy the current test cases for the synchronous driver and ensure that the test cases also pass in the asynchronous version.

MarcinPlaza, 2024-10-14:
Well sad moment, I thought I would finish by this week :( hurricane threw a wrench in my plans. But I believe my async version of the tests is working but for some reason the simulated device refuses to connect so now I am working on fixing that.

MarcinPlaza, 2024-11-18:
I have pushed my attempt to a new branch called [electrical-protocol-async-v2](https://github.com/uf-mil/mil/tree/electrical-protocol-async-v2), for now I have left an "Async_readme.txt" to bring whoever comes next up to speed. Note: I cannot promise the validity of any of the code so assume it might be wrong.