uf-mil/mil #1288: Develop Python interface to sdgps `robosub-ping-solver`

cbrxyz, 2024-10-02:
### What needs to change?

Thanks to the great, amazing, fantastic work of Sylphase, we have a great tool for solving the localization problem with underwater microphones. We will need to develop an interface to this solver, so that we can obtain the locations of the pingers underwater at RobotX.

Relevant docs:  https://docs.sylphase.com/uf/sdgps/preview/docs/html/extract-robosub-pings.html

### How would this task be tested?

1. Ensure that the library can communicate with the sdgps library.
2. Write some tests for the library.
3. Ensure that the library is able to communicate with the Sylphase board when ROS is launched/the boat is launched.

alexoj46, 2024-10-07:
This week I was assigned this task while on pause with the LED tower. To start, I gained access to the Sylphase discord server and GitLab, and I setup a new user for working on this on the workstation. So far I have cloned the uf/sdgps repo, installed sdgps, and successfully built and set-up sdgps on the workstation. I was able to simulate a ping using this software via the command line. I have begun ROS tutorials and research for completing the next steps, which entail writing a python server to receive TCP data from sdgps, and publishing it to the robot with ROS. Separately this week, Cameron taught me more about the general workings of Navigator, especially its software, and I attended today's testing session.

alexoj46, 2024-10-14:
This week, I wrote a Python server receives the simulated ping from sdgps and publishes that message as to a ROS topic. This works by connecting to the sdgps output with TCP, then creating a ROS publisher and retransmitting the message. I plan to tweak the message formatting, as well as clean up the code and change the ROS topic name, in this coming week. I also attended today's testing session.

alexoj46, 2024-10-21:
This week, I opened the PR above (still needs some cleaning up/comments, but wanted it available for testing today). This required integrating the publisher within the mil_passive_sonar package and fixing the message format to be a ROS msg. This also meant building sdgps on my UTM machine and rewriting some things because of the power outage in MALA.  I also attended today's testing session.

PR #1293 (2024-10-28): python publisher for sdgps solution to hydrophone pings
## Description

This PR creates a bash script pipeline for posting the sdgps ping solution message over TCP (fakeping_pipeline.sh creates a fake ping for testing, and pipeline.sh receives real hydrophone pings).  Then, when running the python node ping_publisher.py, it will accept those TCP messages and publish them to the ROS node /ping for future use. These changes are part of the ROS package mil_passive_sonar.

## Screenshot or Video

[image]
Each panel started in order: 
Bottom: Debug messages after running "rosrun mil_passive_sonar f [... cut]
closed 2024-10-28 by cbrxyz