uf-mil/mil #1292: Add tmux status bar modules in a NaviGator tmux config

cbrxyz, 2024-10-18:
### What needs to change?

We should add a custom tmux status bar for NaviGator showing some important values on the boat, such as:

* core temperatures
* battery voltages
* memory/cpu usage

This is the start of a snippet I was working on to print out the battery voltage straight to tmux:
```bash
rostopic echo /odom/pose/pose/position/x | awk '$1 ~ /^[0-9.-]+$/ { printf "%.6f\n", $1 }'
```

### How would this task be tested?

1. Ensure that the config can be launched and that the modules appear.

hoshbosh, 2024-10-28:
I worked on adding custom values to the tmux status bar, I was able to get the code snippet above to show up as a value that updated in the status bar. I did this by making a bash script containing the code snippet in the initial comment, then I made a .tmux.conf file that printed the output of the bash script to the status line. This week I plan on duplicating this to work for the other NaviGator info as listed in the initial comment (i.e. core temps, battery voltages, memory/cpu usage).

hoshbosh, 2024-11-04:
I worked on making bash scripts for reading NaviGator important values, those being core temps, battery voltages, memory and cpu usage. I complied them each into their own bash scripts to be loaded by the .tmux.conf file. I got a few to appear with raw data in the tmux status line. This week I plan on formatting each of them to fit better on the status line and making a pr.