uf-mil/mil #1308: Create `sdgps.service` on NaviGator

cbrxyz, 2024-11-05:
### What needs to change?

Rather than needing to start our GPS each time we turn on NaviGator, it would be better to have a systemctl service to automatically launch the GPS when the computer is started.

### How would this task be tested?

1. Ensure that the GPS is automatically launched when the boat turns on.

thadavale, 2024-11-11:
This week I created a file called start_gps.sh inside Navigator's scripts folder that creates sdgps.service, which runs NaviGator/mission_control/navigator_launch/launch/hardware/gps.launch. Start_gps.sh also runs commands which sets up the service. I also configured the on_boot file inside the scripts folder to run start_gps.sh upon running, which should start the GPS when the Navigator computer starts. I still need to test the code to ensure everything runs smoothly.

thadavale, 2024-11-18:
This week, I began testing my GPS start code but I ran into a few errors getting the service to start when the Navigator simulation starts. Therefore, I will research more into systemctl files and how to run them automatically through Navigator's starting code.

thadavale, 2024-11-24:
I was not able to get much done this week due to exams but I continued researching systemctl files and trying to run them in Navigator's starting code.